wh1ter0se/PowerUp-2018 #18: Figure out how to composite two webcams into one stream for vehicle awareness.

wh1ter0se, 2017-11-24:
Self explanatory. Not sure if we use GRIP or something third party, We can take the approach of just getting an ultra-ultra-ultra-wide cam, but we already have some webcams to start playing with. Plus last year we tried to get one of those and I can't remember why but we never did.

This may actually be as simple as putting two camera streams side by side and combining it into one stream, then physically positioning the cams to align. So yeah, we can probably do this with GRIP.

wh1ter0se, 2017-11-24:
Oh yeah, you guys probably don't know GRIP yet. It's an extension of WPILib for camera code. It's a program with a nice **GUI** _(**G**raphical **U**ser **I**nterface)_ that lets you drag and drop effects and filters and stuff (technical, not like instagram), then converts that media pipeline into Java code for our bot. 

Here's the repo: https://github.com/WPIRoboticsProjects/GRIP

The download is under Getting Started (in the readme.md AKA the main page)

brogan20, 2017-11-26:
Why do we need to combine multiple streams into one? Is there going to be something displayed on the stream that would need the computer awareness stream to be combined with the vision?

wh1ter0se, 2017-11-27:
@brogan20 It's about taking two adjacent webcam streams and combining them in order to double the FOV.
If we have a high FOV, then we can see more of our periphery. If we can see more of our periphery, we can be more aware of what's around the bot. Good question.
[image]

wh1ter0se, 2017-11-30:
@brogan20 If you are going to work on this issue, do your best to document your progress and ask any questions on this thread.

brogan20, 2017-12-01:
GRIP does not seem to have the built in GUI commands to do this (that I could find), but in the org.opencv.core package, the Core class contains a hconcat (horizontal concatenation. vconcat does vertical) method which takes an input Mat (matrix object from opencv) which is two matrices in one (they must have the same number of rows) and an output Mat. The method does everything to attach images. Since OpenCV (and thus GRIP) processes on an image by image basis, the best way to set this up is to combine each photo and then stream it. I didn't play with GRIP too much, so it may be possible in that, but it seems easy enough to do with OpenCV and Java.

wh1ter0se, 2017-12-01:
Good work. Let's write some shell code for it over the weekend. It'll either be in 8.0.3 or 8.0.4, depending on how soon I finish the driftDrive code.

wh1ter0se, 2017-12-01:
I'll actually make a separate issue for this, since this was just a matter of finding the solution, not making it. Good find.

wh1ter0se, 2017-12-01:
@brogan20 Make sure you accept the collaborator invite. It's in whichever email you use for Github. It'll let me assign you to an issue.

closed 2017-12-01 by wh1ter0se