wh1ter0se/PowerUp-2018 #23: Configure a raspberry pi for timelapse (w/ webcam).

wh1ter0se, 2017-11-29:
Dudding and I have been talking, and we think it would be a good idea to set up a timelapse cam that takes pictures of the build room periodically. 

Ideally, this would mean having it take a picture 20 minutes after the start and 20 minutes before the end of each build session after we make a schedule, and then once every x hours on Saturdays. Realistically, it might just mean taking a picture at a certain time (6:30pm?) every day but Sunday.

It would need to be able to run headless (no screen or user interaction) on a pi, and resume the schedule if the pi reboots (in case of power outage).

If you are good with Pi's, PLEASE SPEAK UP. I'll work on this as much as I can, but realistically any help would be good.

meads594, 2017-12-12:
I'm not sure how the webcam is controlled on the pi, but I assume you can snap a picture using the command line. In linux, if you want something to happen at specific times, the way to do this is through the "cron" tool. Basically, you write a script to snap the picture, then use a crontab to run the script at specific times. 

Any thoughts on where the pictures need to end up? I assume they need to make their way off the pi at some point to be "useful".

wh1ter0se, 2017-12-13:
@brogan20 Where are you at with research into this so far?

brogan20, 2017-12-13:
Honestly, I haven't done much at all. Been loaded with schoolwork this week

Tyler-Middendorf, 2017-12-14:
Do you know what pie we might be using because there is a special camera that you can buy for pi’s to do time lapses

Tyler-Middendorf, 2017-12-14:
Check this for how to store the images to a usb if we want that i’m not shure though if we are using a usb or if we want it to do something else. https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=100455

meads594, 2017-12-14:
Do we already have a camera to use for this? Is the plan to use the same cameras we used on the robot last year?

Ideally, once the picture is taken and the image file is saved locally on the pi, it would be nice (if we can get the pi on the network) to copy it over to a machine running a web server. That way you could have a continuously updating web page. I'm not sure what the school network policies are, though. There may be an awful lot of hoops to jump through. The flash drive is a good alternative, if a bit "high latency".

meads594, 2017-12-15:
This webpage looks like a pretty good overview of using a webcam with the pi:

  https://www.raspberrypi.org/documentation/usage/webcams/

There's also a link to documentation on cron for automating it.

brogan20, 2017-12-15:
We already have a camera for the pi - I believe it's the same model as what's on the robot. If we can keep the pi constantly connected to the internet, we could definitely offload all the images. The simplest way would be through [grive](http://www.home-automation-community.com/google-drive-on-raspberry-pi/). Since the school already uses Google Drive for a lot, it shouldn't be any issue to store images there. At this point I think we can set everything up. @wh1ter0se, does build team have any schedule yet? Are we still going to be taking a picture everyday except Sunday regardless of what's going on? And should the images be stored on the pi or Drive?

meads594, 2017-12-15:
Looks promising. Sending the photos to google drive would be great. We'll have to look and see what the options are for grive. Since storage space on the pi is limited, "mirroring" the google drive on the pi may cause complications. It may be better if we could just copy the files directly to google drive.

wh1ter0se, 2017-12-16:
The solution I had in mind would be to set up a *persistent* program takes screenshots from the camera and then saves them with the date and time at every time on its schedule. *Persistence*, or the program continuing to run even after the pi is reset (in case of cables being pulled or a power outage), is really important here as well.

I figured that after we set that up, we could install gdrive on it and change the save directory to the drive folder.

@brogan20 How much have you put on the sim card? Do you think the pi and cam will be ready to mount by Wednesday?

wh1ter0se, 2017-12-16:
Also @meads594 I wouldn't imagine pi storage being a problem. I don't think we established a solid pic schedule yet, but I wouldn't imagine us having more than 300 pics, so even a 4gb would cover us.

brogan20, 2017-12-16:
All that's on the SD card is raspbian. Since all that will be needed once we figure out the schedule and set up a crontab to take pictures at the specified times

wh1ter0se, 2017-12-18:
@brogan20 Try to have it ready by Wednesday. It needs to take pictures at 6:15, 7:30, and 8:45 on M, T, W, TH, and 11am, 12pm, and 1pm on Saturdays. The files needs to be named *3695Powerup-[YYYYMMDDHHSS]*. The devil is in the details. Make sure you test it too, so we know it works with the right cam.

brogan20, 2017-12-20:
Done

SoWaky, 2017-12-20:
Is there a specific reason to take these pics from a camera mounted on the robot?  I would think that the robot would be without power a lot of the time.  If the purpose of taking these pics is to have a nice history of progress in the build room, why not plug in the pi into a wall outlet and mount the camera somewhere in the corner of the room where it can see the whole room?

brogan20, 2017-12-20:
Unless I've extremely misunderstood something, @wh1ter0se didn't mean mounting on the robot, but mounting the pi somewhere in the build room. It'll probably be a setup just like what you described.

SoWaky, 2017-12-20:
Good. I saw the word Robot somewhere and thought you guys were talking
about the camera on the robot.

wh1ter0se, 2017-12-20:
@SoWaky Nope, we're mounting it in the corner of the build room and leaving it running. It gets wall power supply, hence we need to make sure a blackout or accidentally pulled plug won't shut down the program on restart.

brogan20, 2017-12-21:
Project has been finished.

closed 2017-12-21 by brogan20