wh1ter0se/PowerUp-2018 #33: Look into FRC's bot simulator.

wh1ter0se, 2017-12-05:
What do we need to get it working? How feasible is it to get all members to have access to it? Do we need our own CAD files provided or do they provide templates?

brogan20, 2017-12-05:
This link contains all the setup necessary for the bot simulator, titled FRCSim: [https://wpilib.screenstepslive.com/s/currentCS/m/frcsim/l/480159-introduction-to-frcsim](url).
It appears that we'd need our own CAD files for everything, but the play field will likely be provided when the game is revealed.

wh1ter0se, 2017-12-05:
@brogan20 Are there any third party CAD templates? Things like mecanum code would be hard to test with our own CAD, but if there were templates then it would be pretty easy.

brogan20, 2017-12-06:
I don't have Ubuntu set up, so as of now I have no way to tell if any 3rd party template would work, but if under the assumption that they do, a quick search for "FRC mecanum cad" produced many results. The bigger problem seems to be setting this up; Ubuntu is required and FRCSim usually doesn't work on Intel Integrated Graphics so usage in Miller's room (besides on a laptop) would be impossible. This so far is not seeming very feasible if we don't have a strong CAD team.

wh1ter0se, 2017-12-09:
[It looks like they provide two examples](https://wpilib.screenstepslive.com/s/currentCS/m/frcsim)

@brogan20 Could you look into what the steps are between our Autodesk Inventor CAD files and the FRCSim?

brogan20, 2017-12-09:
Turns out Autodesk has their own software, [BXD Simulator](http://bxd.autodesk.com/index.html), with the exact same functionality as FRCSim. Even better, it can be run on Windows so it could be feasible to run it at the school for everyone. There's also many tutorials and models for setting everything up as easily as possible. May be better to use this over FRCSim given Autodesk made Simulator

wh1ter0se, 2017-12-11:
After discussion between AJ and Brogan. I really feel like we should NOT bank on a simulator. It introduces far more potential problems than it fixes. The majority of code problems we have are not things you can simulate, rather things that would happen on a physical bot (janky sensor, PWM crosstalk, bad dimensions, improperly wired electronics, etc.)

In addition to providing very little logistical benefit, keep in mind that we would need fully functional CAD being made with dead accurate precision, that would also need to be fully updated across every single individual part file whenever we make even a slight dimensional update. We have never come anywhere close to that. We've never even just had one complete model of the bot.

Without an intensely detailed simulator so wise that we would consider it magic, a professional CAD team that can model our entire bot with impeccable accuracy and update each part after any slight change, and enterprise-level professional wiring and building to ensure there are no physical problems, using a Sim to code and test everything just isn't feasible.

TL;DR: we don't have enough magic to make this work dependably

closed 2017-12-11 by wh1ter0se