wh1ter0se/PowerUp-2018 #38: Play with our new Neopixels.

wh1ter0se, 2017-12-07:
Based from #24 

The goal is to start creating some cool patterns that we couldn't before via Arduino. If we can, we should see what it would take to control the 'duino via Java on the RoboRio.

meads594, 2017-12-12:
I'm not sure what you mean by "control the Arduino via Java on the RoboRio". 

The arduino is a microcontroller, which runs its own "control loop" software, and has access to various input and output pins that can be used to control the program flow. The RoboRio is a (bigger, fancier) microcontroller that essentially operates in exactly the same manner. 

My first thought on "communication" between the Arduino and the RoboRio would be to attach an analog or digital output pin (or pins) on the RoboRio to an input pin (or pins) on the Arduino.

So, as an example, if the robot is in autonomous mode, you could have the (java) code on the RoboRio set some pin value. The Arduino would see this and could then flash the LEDs in a particular pattern. After autonomous, the pin value could be changed, then the Arduino could change the LED pattern (for example). You could also imagine the RoboRio setting one of these pin values when something specific happens (like a pneumatic solenoid fires). The Arduino could then flash the lights (or something like that).

wh1ter0se, 2017-12-12:
By "control the Arduino via Java on the RoboRio", I mean control what the 'duino is telling the lights to do, via Java on the Rio. I was essentially just saying we need to figure out how to send those signals to the 'duino and properly use those values on the 'duino.

The challenge isn't so much the concept so much as it is the specifics. We need to figure out what Java code we need to manually change the voltage of an output pin on the Rio, and then it shouldn't be too hard to convert that 'duino input voltage to a variable. 

This issue is really more focused on just figuring out how to control our Neopixels via Arduino, though. From there we can start working out the specifics of how to wire it for variable communication.

Tyler-Middendorf, 2017-12-13:
as far as controlling the neopixel lights with the arduino uno I found [someone](https://github.com/adafruit/Adafruit_NeoPixel) who has done the most of the code we would need.
They also have in the examples folder rgbw lights so we can use the white lights!

wh1ter0se, 2017-12-16:
I'm gonna add a second issue for the Arduino-Rio communications and leave this issue for playing with different patterns only on the duino. 

@Tyler-Middendorf Let's use that to base our patterns off of. I was hoping we could start playing with parameter-based functions, like a pulse that flies along the strip based on parameters speed and length, or color-based sorting to display what range a value is in. Things like this, but not limited to this.

wh1ter0se, 2017-12-17:
Don't forget we also have the uberguide:
https://learn.adafruit.com/adafruit-neopixel-uberguide?view=all

meads594, 2018-01-17:
I played around with the NeoPixels a bit last night. They are working - I used the "resistor chain" and the Arduino. I was powering the +5V directly from the Arduino, and it seemed to supply enough current. (Note - one end of the strand has a male connector and the other has a female connector.  It doesn't matter which end you attach to and you can plug a wire directly into the female connector. It's not a permanent solution, but works good enough for testing.)

I was able to run a couple of the sample sketches from the AdaFruit library (blinky lights!) and also fiddled a bit with turning individual LEDs on/off and changing the colors. 

I think the next step is to figure out how the NeoPixels will be used, and what control signals we'll need.

closed 2018-01-19 by wh1ter0se