wh1ter0se/PowerUp-2018 #45: Figure out communications between the Rio and Arduino for Neopixels.

wh1ter0se, 2017-12-16:
Based off #38.

The plan is to use the output pins on the Rio and the input pins on the duino. 

This is really just matter of figuring out how to manually change the voltage of a pin in WPILib.

Nathan-2018, 2017-12-17:
would we need resistors because i don't think the duino can take more than 1 amp per pin?

meads594, 2017-12-18:
[375275a.pdf](https://github.com/wh1ter0se/PowerUp-2018/files/1569041/375275a.pdf)
I've attached the specification sheet for the RoboRio. 

We'll have to decide if we want to use analog or digital pins for the communication between the Arduino and the RoboRio. Digital is probably easier, but may require multiple pins, depending on what we want to do with it. In principle, if we used an analog pin, we could have different voltage levels mean different things. (Although this has its own complications.) 

But, in general, you are right. If we use an analog pin then we will have to watch the amount of current to make sure we don't fry the arduino. (The RoboRio documentation says it can only drive 3mA of current, so it may not be a problem.)

Tyler-Middendorf, 2018-01-09:
I found an amaizing string on chiefdelphi talking specifically about this. It talks about how we should approach this communication.

https://www.chiefdelphi.com/forums/showthread.php?t=132572

brogan20, 2018-01-28:
Yeah. We got the roborio sending signals to the Arduino. Just have to solve getting multiple parameters in one go now, so issue #73.

closed 2018-01-28 by brogan20