wh1ter0se/PowerUp-2018 #48: Add a function to broadcast field setup.

wh1ter0se, 2018-01-08:
[How to access field setup.](http://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/826278-2018-game-data-details)

DriverStation.getInstance().getGameSpecificMessage() will return a string [L/R][L/R][L/R], in the order of [our scale][switch][opponent's scale].

I'm debating between declaring it in constants or in one of the autonomous classes.

brogan20, 2018-01-08:
I'd put it in an autonomous class myself

wh1ter0se, 2018-01-08:
Just realized I didn't commit last night. I added a direction enum [left,right], and a field class in the util folder, with 3 get methods for each obstacle. 

Revision: I didn't commit yet because I need to move WPILib18 over first, it relies on a new method.

wh1ter0se, 2018-01-08:
@brogan20 I'll try to take care of it after the meeting tonight.

wh1ter0se, 2018-01-09:
Just added it and updated wpilib

closed 2018-01-09 by wh1ter0se