wh1ter0se/PowerUp-2018 #75: Create static patterns for the LEDs.

wh1ter0se, 2018-01-28:
***PLEASE ADD TO THIS LIST OR CREATE PATTERNS AS MUCH AS YOU WOULD LIKE TO***

We pretty much want to write a library of parameter-free patterns to use whenever we want.

Team color dependent patterns
- [x] Solid team color, with chasing pulse of intensity
- [x] Solid team color, with intense white pulse (seriously bright)
- [ ] Bouncing chase (chase, but reverses direction when it hits the end)
- [ ] Bouncing chase with two points
- [ ] Breathing (solid intensity, intensity goes up and down)
- [ ] Building chase, of White-Team Color-White-Team Color pattern (when I say building chase, I mean a turned off strip that turns on as the chase goes through and stays on)
- [ ] Alternating White-Team color flashing pattern
- [ ] Alternating Off-Team color flashing pattern

Independent color patterns
- [x] Slow, fast, and sonic fast rainbow fade
- [ ] Red-Blue alternating chase (probably good for things like demonstrations)
- [ ] Rainbow long-chasing (chase that adjusts color with each one, carriages being 30 LEDs long)
- [ ] 3-pixel chase of solid red, yellow, and blue (or R, G, B?) LEDs in sequence

Found some cool ideas in this vid here: (feel free to post some that you find if there are any other patterns)
https://www.youtube.com/watch?v=pQwgZwrXfhc

wh1ter0se, 2018-01-28:
Definitely one of our more interesting issues.

brogan20, 2018-01-28:
This is definitely one anyone could work on. If anyone wants to see examples for this check out the code in the Edward branch in the ArduinoCode folder and look at the solid color and light chase methods for examples of how they work.