wh1ter0se/PowerUp-2018 #81: Create the GRIP pipeline for CyborgCommandAscend.

wh1ter0se, 2018-01-30:
This can be calibrated later, but we want to have it ready as soon as possible. It needs to locate the two pieces of reflective tape using a webcam mounted on the candy cane, and then quickly approach (adjusting left and right as needed), drop the hook on the bar, and then ascend.

meads594, 2018-01-30:
Remember that the reflective tape is on the switch and not the scale. So, it won't help for locating the bar. There is a pattern on the side of the scale. I don't know how hard (or easy) it would be to pick up on it. 
[image]

brogan20, 2018-01-31:
Mr. Eads is right: The rungs have no reflective tape on them so we can't detect them through vision, or at least not as easily as with reflective tape. If we really want to detect the rungs we'll likely need another method or to do nearly all detection at competition.

closed 2018-02-11 by brogan20