wh1ter0se/PowerUp-2018 #99: Figure out why the flywheels aren't receiving run values.

wh1ter0se, 2018-02-13:


wh1ter0se, 2018-02-13:
AHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHH

No bugs found so far. Talons won't display on the web config so we're going to try to fix that first.

brogan20, 2018-02-14:
More info/instructions: https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md#installing-phoenix-framework-onto-your-frc-robot

Phoenix framework download: http://www.ctr-electronics.com/control-system/hro.html#product_tabs_technical_resources

Good luck!

wh1ter0se, 2018-02-15:
"A workaround bypassing any command structure has been implemented and works. Subject to change if the command issue is figured out"
-Probably Brogan

wh1ter0se, 2018-02-15:
@brogan20 why are you violently opposed to just commenting with your own account

wh1ter0se, 2018-02-15:
Also we can put this on the backburner for now, but we should figure this out and make a clean solution for it in the next couple weeks. Trust me, we do _**not**_ want to walk into competition with inexplicable patches being heavily relied on in our code. We get enough inexplicable errors as is.

wh1ter0se, 2018-02-18:
Closing this for now because we have higher priorities than fixing that patch.

closed 2018-02-18 by wh1ter0se