wh1ter0se/PowerUp-2018 #102: Add warnings and issues throughout the code to send.

wh1ter0se, 2018-02-15:
I think sending status updates for subsystem instantiation, command calls, and CyborgCommand progress to the driver station could be really useful for driver awareness. Could potentially make program debugging even easier too.

meads594, 2018-02-15:
I'm not sure what is available in the driver station interface, but most logging tools have a lot of flexibility. For example, you can typically log an "error", a "warning", or an "info" message. Sometimes, it will also allow for an integer "level" associated with each message. This makes displaying (or searching through) log message very configurable and easy.

wh1ter0se, 2018-02-18:
Good enough.

closed 2018-02-19 by wh1ter0se