wh1ter0se/PowerUp-2018 #123: Tune rudimentary PIDF config so we can test autonomous (and figure out what F is???)

wh1ter0se, 2018-02-20:


brogan20, 2018-02-20:
Chapter 12 of the Talon SRX Software Reference Manual is a walk-through of tuning everything

wh1ter0se, 2018-02-20:
@brogan20 We need to figure out why the getAndSets for PIDF aren't sending though. We can't tune if we can't configure.

brogan20, 2018-02-20:
There always is the web interface. The software manual says that works just as well.

wh1ter0se, 2018-02-20:
@brogan20 Wouldn't we have to set it for every talon, every time if we did that? Plus we would be tethered in testing

brogan20, 2018-02-20:
I mean we'd have to either way. Since the bot is pulling to the right we're going to need different values for each side

brogan20, 2018-02-20:
For the F value, it seems we'll have to be tethered to calculate it too

closed 2018-03-01 by wh1ter0se