wh1ter0se/PowerUp-2018 #126: Research the gyroscope in the RoboRio for the anti-tip function.

wh1ter0se, 2018-02-20:
*START BY PUBLISHING THE GYRO VALUES TO THE DASH*

We pretty much need to figure out how to override user control and accelerate *into* wherever we're tipping *until* we are stable again. This is so we can abuse the hell out of it. 

We may not necessarily use it, but I'd rather have it and not need it than vice versa.

brogan20, 2018-02-20:
We don't have a gyroscope. By using the accelerometer we're calculating the angle we're currently at. Just want to make sure the distinction is clear. We already do get the Y-angle with one of the methods in SubsystemDrive, so putting that on the dash shouldn't be too hard

wh1ter0se, 2018-02-20:
@brogan20 Think you could do it tonight?

brogan20, 2018-02-20:
I suppose so. Do you want all angles and where would you want it set from so it's constantly updated?

wh1ter0se, 2018-02-20:
@brogan20 anywhere it would run iterative (called every time the command executes/the method runs). and just the Y should be fine. If you're adding things to dash rn, would you mind putting booleans on the dash to update whether the bot is in reverse and if its docking on there?

brogan20, 2018-02-20:
I'll stuff it all in ManualCommandDrive

wh1ter0se, 2018-02-20:
@brogan20 oof

brogan20, 2018-02-20:
Added the booleans and Y value to dash (part of the optimize all the things commit)

wh1ter0se, 2018-02-20:
@brogan20 Good good

brogan20, 2018-02-20:
The getYAngle method now reports our current tilt in degrees and puts it on smartdash. All set.

closed 2018-02-20 by brogan20