wh1ter0se/PowerUp-2018 #174: Get rid of the hiccup in dualPID.

wh1ter0se, 2018-03-13:
We get a consistent hiccup when running left switch and left scale in dualPID around ~60in into the command. We have not tested this with right side yet, so running that might give us some insight. 

The weird thing is, nothing in changed code would seem to cause this. I think the best approach, after testing the right side, is to manipulate the PIDF, velocity, and acceleration values (even in ways that don't make sense) to see if one is responsible for the weird changes.

It happens at a consistent distance (seemingly), not just a certain velocity, and it's the same distance for either the switch or the scale. 

What other variables in the dash affect the drive train, besides PIDF, velocity, and acceleration?

What is the specific distance that the stutter happens at?

Will we get the stutter if we run it at a lower speed?

@brogan20 Feel free to spitball some theories here

wh1ter0se, 2018-03-13:
What is the velocity at when the robot hiccups? Could it be some sort of velocity limit?

If we run all the way to enemy switch, will the hiccup be in the same spot? If so, why is the hiccup in an absolute position and not relative to the end goal, considering the goal of the distance loop never resets?

meads594, 2018-03-13:
Is it possible to roll everything back (both the code and the PIDF values) to what is was before you began all the tuning yesterday. Starting from a known point (hopefully without the hiccup) may make it easier to figure out where the hiccup comes in. 

Is it possible to pull any data (versus time), either from the talon or from the rio, to see what talon is actually doing? If we knew the set point and current values (versus time), it should be possible to figure out what the PIDF control loop is doing. Without knowing this, changing/tuning the PIDF values is really just guesswork.

closed 2018-03-16 by wh1ter0se